sensepost/USaBUSe #1: Gerbers

dwhacks, 2016-08-18:
I was thinking of trying to build a board but there is no board outline file in the gerbers folder, at least not one oshpark recognized.

singe, 2016-08-18:
We'll get on that.

RoganDawes, 2016-08-19:
Hi @dwhacks,

Our designer has informed us that the board outline provided was actually for two boards on a single PCB. He will provide an outline of a single board shortly, I hope to have it up by Monday.

RoganDawes, 2016-12-09:
Updated Gerbers have been added to the repository.

closed 2016-12-09 by RoganDawes

dwhacks, 2017-03-31:
I have tried adding the Gerbers to Oshpark. First it seems the outline is still for two boards, and after this it is for one board plus a blank section that should be between the two board version. 

I would love to build one of these, so maybe you could get a working board on Oshpark for people like me who just want One or Two could order. 

Thanks!