sensepost/USaBUSe #9: same design like USaBUSe

h4kb4n, 2016-08-22:
https://github.com/RadioWar/FireGoose 
we open source now

LucaBongiorni, 2016-08-22:
English is usually much more appreciated than Chinese language ;)

h4kb4n, 2016-08-22:
no problem we can translate to english

LucaBongiorni, 2016-08-22:
That's cool! Thanks!
P.S. I have seen that Firegoose is on sale on Taobao. Maybe it could be put on Aliexpress/Ebay?
Because Taobao is quite closed for non-chinese buyers. And using a broker is not very comfortable. ;]

h4kb4n, 2016-08-22:
Sorry man, already sold out but i can send one pcs for you ,  can dm me your address

LucaBongiorni, 2016-08-22:
Too bad! Keep me in consideration for the next batch then ;]

RoganDawes, 2016-08-22:
Hi,

This looks interesting - it's definitely a good idea to have a more powerful processor dealing with the USB side of things, for best performance! I looked at the BOM on your github site, but was not able to see which wireless module you are using?

Rogan

closed 2016-10-31 by RoganDawes

h4kb4n, 2016-11-29:
i can give you wifi module info

RoganDawes, 2016-11-29:
Thank you. I look forward to seeing what you are using.

h4kb4n, 2016-11-30:
FSEUMH1D1-08G
WM7601WU-YM(Wireless Module)

we dont want to update any more ,so if you want to keep going ,  we will very happy ! thx

RoganDawes, 2016-11-30:
I do intend to develop this further, mainly to improve performance, but also to add new functions.

h4kb4n, 2016-12-01:
great,if you want to help or anything tell me~ i will support it