sensepost/USaBUSe #11: Trouble flashing firmware

Relys, 2016-08-25:
Hi,

I purchased several Cactus Micro Rev2's directly from https://www.tindie.com/products/AprilBrother/cactus-micro-rev2-arduino-compatible-plus-esp8266/

I'm attempting to flash them but am recieving this error "Failed to connect to ESP8266"

```
root@kali:~/USaBUSe# esp-vnc/flash_esp ~/Downloads/user1.bin ~/Downloads/KeyboardMouseGeneric.hex 
Waiting for the device to go away ...........
Waiting for it to come back ..
Got it!
Connecting to programmer: .
Found programmer: Id = "ARDUINO"; type = d
    Software Version = s.c; Hardware Version = a.n
avrdude: error: buffered memory access not supported. Maybe it isn't
a butterfly/AVR109 but a AVR910 device?
Connecting...

A fatal error occurred: Failed to connect to ESP8266
Flashing /root/Downloads/KeyboardMouseGeneric.hex to the atmega32u4, press reset to begin
Waiting for the device to go away ........
Waiting for it to come back ..
Got it!
Connecting to programmer: .
Found programmer: Id = "ARDUINO"; type = d
    Software Version = s.c; Hardware Version = a.n
avrdude: error: buffered memory access not supported. Maybe it isn't
a butterfly/AVR109 but a AVR910 device?
```

Relys, 2016-08-26:
Ok, first I grabbed and extracted the ESP8266 SDK to get esp_iot_sdk_v1.5.1:
```
  $ wget --content-disposition "http://bbs.espressif.com/download/file.php?id=1046"
  $ unzip ESP8266_NONOS_SDK_V1.5.1_16_01_08.zip
```

Next, I connected the Cactus Micro Rev2 and pressed reset twice to enter bootloader programming mode and run the flashing script:
```
root@kali:~/USaBUSe# esp-vnc/flash_esp ~/Downloads/user1.bin ~/Downloads/KeyboardMouseGeneric.hex 
Waiting for the device to go away ...
Waiting for it to come back .
Got it!
Connecting to programmer: .
Found programmer: Id = "CATERIN"; type = S
    Software Version = 1.0; No Hardware Version given.
Programmer supports auto addr increment.
Programmer supports buffered memory access with buffersize=128 bytes.

Programmer supports the following devices:
    Device code: 0x44

Connecting...
Erasing flash...
Wrote 3072 bytes at 0x00000000 in 0.2 seconds (125.9 kbit/s)...
Erasing flash...
Wrote 279808 bytes at 0x00001000 in 18.5 seconds (120.8 kbit/s)...
Erasing flash...
Wrote 4096 bytes at 0x0007e000 in 0.3 seconds (127.0 kbit/s)...

Leaving...
```

Once the ESP6288 is programmed it will say "press reset to begin" to program the second payload used for USB HID attack on the atmega32u4. If I just press it once I get:
```
Flashing /root/Downloads/KeyboardMouseGeneric.hex to the atmega32u4, press reset to begin
Waiting for the device to go away .....
Waiting for it to come back ..
Got it!
Connecting to programmer: .avrdude: butterfly_recv(): programmer is not responding
```
 If I press the reset button twice (like for the first stage flash) I get:
```
Flashing /root/Downloads/KeyboardMouseGeneric.hex to the atmega32u4, press reset to begin
Waiting for the device to go away ..
Waiting for it to come back ..
Got it!
Connecting to programmer: .
Found programmer: Id = "CATERIN"; type = S
    Software Version = 1.0; No Hardware Version given.
Programmer supports auto addr increment.
Programmer supports buffered memory access with buffersize=128 bytes.

Programmer supports the following devices:
    Device code: 0x44
```

However, I am not seeing the SSID the ESP8266 should be broadcasting when I power on the device. Also, Windows complains that the "USB Device is not recognized".

RoganDawes, 2016-08-30:
OK, you are doing everything right, as I have been doing myself. The only thing that you might have to do (if you can find the ESP access point) is set the baud rate on the serial port to 250 000 baud.

However, if the ESP SSID is not showing up, I'm really not sure why that is happening. The log of your flashing shows the expected output, and yes, you do need to press the reset button twice to get into the bootloader, not just once.

Could you possibly show the output of lsusb -v once the device has been programmed by your Kali instance? That might indicate where the problem lies.

RoganDawes, 2016-12-09:
Closing due to lack of feedback.

closed 2016-12-09 by RoganDawes