sensepost/USaBUSe #13: Ideas

DeveloppSoft, 2016-12-03:
Hey sensepost,

I am here to make some suggestions regarding the typing process.

You are currently using a big VNC server running on the esp. I can see there some problems:
- The project works only with one type of keyboard (what if your target has a Chinese keyboard?)
- It may use a lot of esp's ressources

Here is my suggestions:
- Do not use a vnc server, replace it with a tiny server sending received codes to the avr
- The client sends the correct codes to the esp

Feel free to contact me if you want to discuss on that, or close the issue if you do not want :smirk:

RoganDawes, 2016-12-09:
I agree that there are potentially problems if the keyboard mapping is different to the actual mapping in use. I wonder if there is not a better way of approaching it, perhaps by uploading a key mapping to use when translating VNC keys to key presses.
It certainly is possible to implement an additional server that simply receives mouse movements and keystrokes generated by the attacker (that's how I prototyped it initially!), but that then means an additional attacker-side component to generate these events.

DeveloppSoft, 2016-12-09:
@RoganDawes I fully agree with you.
Uploading a new keymap is a nice idea!
Actually, I think that having an other attacker-side component is better (essentially for performances and compatibilities issues).
But since it is your project, you decide!
By the way, I will consider shipping a cactus r2 micro for Christmas in order to (maybe) send some pull requests.

closed 2018-08-22 by DeveloppSoft