sensepost/USaBUSe #16: wifi configuration

jonathan-annett, 2017-06-15:
I'm looking at using this tech for a security related application, which is neither pen-testing nor "attack" related, and as such i am hoping to evaluate it for my needs.

I'm waiting for my device to arrive, and have been looking at the various github listings that are apparently compatible with the hardware i have ordered (https://www.aliexpress.com/item/Cactus-Micro-compatible-board-plus-WIFI-chip-esp8266-for-atmega32u4/32318391529.html)

I'm not really prepared to play around with the device unless i can change the default ssid and password, preferably programatically by sending a message to the device over the network, or at the very least at compile time)

one of the other repositories has some code for setting these credentials (https://github.com/spacehuhn/wifi_ducky/blob/master/esp8266_wifi_duck/esp8266_wifi_duck.ino) but i am more impressed with the general approach taken in the USaBUSe project than the wifiducky project, however i can't find anywhere the wifi information is setup in USaBUSe.

am i misunderstanding the scope of this project? is this project not appropriate for the hardware linked to in my first paragraph?

RoganDawes, 2017-06-15:
The USaBUSe project is compatible with the Cactus WHID, so you should be fine there.

The SSID and password can be changed online via a web interface (see esp-link documentation, as the ESP8266-side firmware is based on that project.) or it can be compiled into the firmware at compile time.

RoganDawes, 2017-06-15:
May I ask what your intended usecase is? Perhaps I can give some guidance on whether it will work as is, or whether you will need to make some changes.

jonathan-annett, 2017-06-16:
Sure.

I'm basically going to make a personal, offline "lastpass" style server that lives in an effectively air gapped armbian based SOC device. I'll probably mess with the protocol a bit and use a rotating random ssid/psk between the storage device and the USB key. I'll the  spin up a separate wifi on the storage device that has a more compressive web Gui that a smartphone can talk to, using a  standard browser. (Again air gapped in the sense that device won't go anywhere near other networks.)

This way I can store credentials for any thing I need to log into, without fear it might be compromised en mass.

Well that's the theory 

I'm sure there are holes in my design but there are definite holes in lastpass model which does not support (for example) bios boot passwords or machine login credentials for workstations

closed 2017-10-09 by RoganDawes