sensepost/USaBUSe #20: Problem with the avr firmware

michel950911, 2024-01-29:
Hi, i have a problem with the firmware, i flashed the avr 32u4 with avrdude, but when i plug it in my pc that's not recognise the device,  and no have com port to reflash it, i used windows10 22h2 and the firmware i flashed Is the KeyboardMouseGeneric.hex provide un the release section. Help me please

michel950911, 2024-01-29:
If that help un somethink, in device administrator saya error on request of device descriptor

RoganDawes, 2024-01-29:
The sequence of flashing is quite important.

First, flash the Program_ESP firmware onto the avr. If there is no serial port, you need to double tap the reset button to get it into the bootloader mode.

Then, flash the ESP8266 firmware onto the ESP, making use of the serial port presented by the avr.

Finally, flash the KeyboardMouseGeneric firmware onto the avr.

If the ESP8266 firmware is not properly flashed, the avr will not enumerate as any USB device, as it waits to see a certain sequence of characters from the ESP8266 before doing so.

Which specific hardware are you trying to flash this onto?

RoganDawes, 2024-01-29:
Hi,

The wiring between the 32u4 and the esp8266 is as per the cactus micro rev2: https://wiki.aprbrother.com/en/Cactus_Micro_Rev2_Hookup_Guide.html#the-pinout

The only pins needed by the USaBUSe firmware are the Rx/Tx pins for UART comms between the 32u4 and the ESP, and the GPIO0 and ENABLE (CH_PD) for programming the ESP8266 (GPIO0 puts the ESP into serial bootloader mode).

There *might* be a problem if you are doing your own wiring, as there is some autodetect code in the 32u4 that attempts to identify the Cactus Micro Rev2, or the BlackBox-produced custom hardware that was made for the Defcon talk. Unfortunately, IIRC, the CH_PD and GPIO0 wires were crossed. IIRC, the code attempts to identify an LED present on the BlackBox hardware, and defaults to the Cactus Micro settings.

Of course, you can always just program the NodeMCU yourself without the 32u4 in the way. I do suggest that you connect the CH_PD pin, so that the booting of the NodeMCU happens after the 32u4 is ready to read from the UART, and does not miss the magic string that it is waiting to receive from the ESP.

RoganDawes, 2024-02-12:
Did the above comments help you? Can I close this issue?

closed 2024-02-15 by RoganDawes